arduino-libraries/ArduinoLowPower #3: nRF52 Arduino core

kriswiner, 2017-08-22:
I am using Mistry's Arduino core and a custom nRF52 dev board and I tried to compile one of the simple timed low power examples and I am getting a couple of compile errors:

C:\Users\kris\AppData\Local\Temp\arduino_build_473580\sketch\serial_example.ino.cpp.o: In function `ArduinoLowPowerClass::sleep(int)':

C:\Users\kris\AppData\Local\Temp\arduino_build_473580\sketch/ArduinoLowPower.h:43: undefined reference to `LowPower'

C:\Users\kris\AppData\Local\Temp\arduino_build_473580\sketch/ArduinoLowPower.h:43: undefined reference to `ArduinoLowPowerClass::sleep(unsigned long)'

looks like I am missing the LowPower C module; where am I supposed to get this?

facchinm, 2017-09-04:
At the moment, the library is only compatible with Primo (and Primo Core) nRF52 based boards. 
@sandeepmistry 's core is fundamentally different and probably doesn't define `ARDUINO_ARCH_NRF52` , which is needed for the library to compile correctly.
Could you provide the exact version of both the core and the library you are targeting? Thanks!

wenbozzz, 2020-02-28:
Hi, I believe I have a similar issue here

I'm using the [ARDUINO NANO 33 BLE](https://store.arduino.cc/usa/nano-33-ble) which have a `nRF52840 ` microcontroller. 
I'm using the libary verson 1.2.1
Here is my code (file `test_low_power.ino`) to test the low power sleep:
```
#include "ArduinoLowPower.h"

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);

  // wait for a second
  delay(1000);

  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);

   // wait for a second
  LowPower.deepSleep(1000);
}
```

However this doesn't compile:
```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\wenbo\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\wenbo\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\wenbo\OneDrive\Documents\Arduino\libraries -fqbn=arduino:mbed:nano33ble -ide-version=10810 -build-path C:\Users\wenbo\AppData\Local\Temp\arduino_build_589190 -warnings=default -build-cache C:\Users\wenbo\AppData\Local\Temp\arduino_cache_69019 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.openocd.path=C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\wenbo\workspace\arduino_sketch\test_low_power\test_low_power.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\wenbo\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\wenbo\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\wenbo\OneDrive\Documents\Arduino\libraries -fqbn=arduino:mbed:nano33ble -ide-version=10810 -build-path C:\Users\wenbo\AppData\Local\Temp\arduino_build_589190 -warnings=default -build-cache C:\Users\wenbo\AppData\Local\Temp\arduino_cache_69019 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.openocd.path=C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\wenbo\workspace\arduino_sketch\test_low_power\test_low_power.ino
Using board 'nano33ble' from platform in folder: C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4
Using core 'arduino' from platform in folder: C:\Users\wenbo\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.4
Detecting libraries used...
"C:\\Users\\wenbo\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g -Os -nostdlib "@C:\\Users\\wenbo\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\wenbo\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED "-IC:\\Users\\wenbo\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "-IC:\\Users\\wenbo\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE" "-IC:\\Users\\wenbo\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\wenbo\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "@C:\\Users\\wenbo\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/includes.txt" "C:\\Users\\wenbo\\AppData\\Local\\Temp\\arduino_build_589190\\sketch\\test_low_power.ino.cpp" -o nul
Alternatives for ArduinoLowPower.h: [Arduino_Low_Power@1.2.1]
ResolveLibrary(ArduinoLowPower.h)
  -> candidates: [Arduino_Low_Power@1.2.1]
[... cut ...]
C:\Users\wenbo\OneDrive\Documents\Arduino\libraries\Arduino_Low_Power\src/ArduinoLowPower.h:49: undefined reference to `ArduinoLowPowerClass::deepSleep(unsigned long)'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "ArduinoLowPower.h"
 Used: C:\Users\wenbo\OneDrive\Documents\Arduino\libraries\Arduino_Low_Power
Using library Arduino_Low_Power at version 1.2.1 in folder: C:\Users\wenbo\OneDrive\Documents\Arduino\libraries\Arduino_Low_Power 
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```
any help would be appreciated

facchinm, 2020-02-28:
On Nano33BLE you don't need any explicit power saving trick since all `delay`s will sleep; when you select an example from this library it should appear under "UNSUPPORTED" for this reason (hence the line `WARNING: library Arduino_Low_Power claims to run on samd, nrf52 architecture(s) and may be incompatible with your current board which runs on mbed architecture(s).` )

tangtangdn, 2020-08-06:
Hi, I also use Arduino Nano 33 BLE and I would like to make the system to deep sleep to save more power. Is any other library available to achieve low power consumption? Thank you.

ezechiele2517, 2022-05-16:
Hi, I imagine that the same issue there is on Nicla Sense ME. I got the same errors!
[image]

How can it be solved?

akostadinov, 2024-05-10:
The other option for Nano 33 BLE seems to be `NRF_POWER->SYSTEMOFF = 1;` with some more limited wake up options.

See
https://forum.arduino.cc/t/nano33-ble-how-should-i-go-about-reducing-current-draw/637270
https://forum.arduino.cc/t/put-nano-33-ble-into-power-on-sleep-mode/1169463