arduino-libraries/ArduinoLowPower #5: Not working on SAMD

disk91, 2018-06-10:
The following code never wakeup after the first sleep call (led always on)
#include <ArduinoLowPower.h>
void setup() {
  // put your setup code here, to run once:
  pinMode(6,OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(6,HIGH);
  LowPower.sleep(500);
  digitalWrite(6,LOW);
  LowPower.sleep(500);
}

disk91, 2018-06-10:
tested on Arduino MKRFOX1200

disk91, 2018-06-10:
To be more precise : it works correctly if I remove power supply and restore power it. But it stops working when the device has been programmed from USB.

facchinm, 2018-06-11:
Hi @disk91 ,
this may depend on your operating system/USB connection; the code path in case of USB connection is quite different from the normal one.
Could you try reverting https://github.com/arduino-libraries/ArduinoLowPower/commit/8cf4c737ddb755e76c20407df3d70cc507bd9651 and see if the behaviour is the expected one (in your setup)?

sjernigan, 2018-06-13:
We are seeing a similar problem.  We also see that sometimes the first sleep call almost immediately returns.
Current Setup: Feather M0, using Mac to program/power, reverted the commit above as asked--
When powered from a battery (no data signal), the first call immediately returns, subsequent calls work correctly.  Same for power by mac after unplugging. However, after you reprogram, the first call will never return.

warriorfenixSM, 2018-06-26:
I have the same problem

sslupsky, 2018-09-13:
I think I have this problem as well.  I think it may have something to do with the rtc.  Recently I added an rtc.begin() in my setup() and this has stopped the OP’s issue from happening.

sslupsky, 2018-09-14:
I found another reference to this issue:

https://forum.arduino.cc/index.php?topic=499101.0

Gormd reported that using LowPower.attachInterruptWakeup() also cleared this up.  Seem's to me there is a bug here related to declaration or construction of the RTC object?

DanielRIOT, 2020-04-07:
You could call  
 SerialUSB.end(); just before calling  LowPower.sleep();
to stop the USB interrupts from happening and continuously waking the system

nekuneko, 2020-07-26:
Is there any reason to use `USBDevice.standby();`  call ? I think it doesn't work in the propper way and in order to avoid issues, I would leave `sleep()` function simply as this:
```
void ArduinoLowPowerClass::sleep() {
	USBDevice.detach();
	SCB->SCR |= SCB_SCR_SLEEPDEEP_Msk;
	__DSB();
	__WFI();
       USBDevice.attach();
}
```