arduino-libraries/ArduinoLowPower #7: Serial not wotking after wakeup

tatulea, 2018-09-24:
Hi,

I am using a button to put Arduino in sleep mode and wake it up, but the Serial is not working anymore after it is waking up. What should I do?

`
#include "ArduinoLowPower.h"

volatile bool just_wakeup = true;

// Pin used to trigger a wakeup
const int pin = 0;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  pinMode(pin, INPUT_PULLUP);
}

void loop() {
  if (just_wakeup) {
    just_wakeup = false;
    delay(3000);
    Serial.begin(115200);
  }
  digitalWrite(LED_BUILTIN, HIGH);
  delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);
  Serial.println("test");

  if (digitalRead(pin) == LOW) {
    Serial.end();
    delay(500);
    LowPower.attachInterruptWakeup(pin, wakeup_handler, FALLING );
    LowPower.deepSleep();
  }
}

void wakeup_handler() {
  just_wakeup = true;
  detachInterrupt(pin);
}
`

facchinm, 2018-09-24:
Hi @tatulea, wakeup/standby with the USB connected is super tricky since it depends on the operating system/USB controller the board is connected to. 
Would you mind sharing your PC or Mac configuration (both software and hardware) so I can try reproducing it? Thanks

grezco, 2018-09-29:
Hello,

I have the same problem with MKR WiFi1010 (and MKR1000) - Win10

durasno, 2018-09-29:
Hello,

I also have the same problem with the Arduino WiFi MKR 1010 and MacBook Pro running OS X Yosemite.

facchinm, 2018-10-01:
@tatulea @grezco @durasno can you take a look at https://github.com/arduino-libraries/ArduinoLowPower/pull/8 ? Let me know if the proposed behaviour could work for you.

grezco, 2018-10-01:
Hi @facchinm

facchinm, 2018-10-01:
@grezco which operating system/PC are you using?

grezco, 2018-10-01:
Windows 7 and Windows 10

durasno, 2018-10-06:
> @tatulea @grezco @durasno can you take a look at #8 ? Let me know if the proposed behaviour could work for you.

Hi @facchinm,

The below script (based on the [TimedWakeup](https://github.com/arduino-libraries/ArduinoLowPower/tree/master/examples/TimedWakeup) example) seems to be working relatively well (after brief testing with Arduino WiFi MKR 1010, MacBook Pro + OS X Yosemite) with the changes from #8. The added delay is from a [LowPowerLab/LowPower SAMD21 example sketch](https://github.com/LowPowerLab/LowPower/tree/master/Examples/standbyExternalInterruptSAMD21).

```#include <ArduinoLowPower.h>

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  // Uncomment this function if you wish to attach function dummy when RTC wakes up the chip
  LowPower.attachInterruptWakeup(RTC_ALARM_WAKEUP, dummy, CHANGE);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);
  
  while(!SerialUSB);
  delay(1000); 
  SerialUSB.println("Just woke up!");
  
  // Triggers a 2000 ms sleep (the device will be woken up only by the registered wakeup sources and by internal RTC)
  // The power consumption of the chip will drop consistently
  LowPower.sleep(2000);
}

void dummy() {
  // This function will be called once on device wakeup
  // You can do some little operations here (like changing variables which will be used in the loop)
  // Remember to avoid calling delay() and long running functions since this functions executes in interrupt context
}

grezco, 2018-10-06:
Hi @durasno,

The same script testing with WiFi MKR 1010 and **Windows10** with the changes from #8 Don't work for me.

durasno, 2018-10-06:
Hi @grezco,

If it helps I wait for the led to flash and then repeatedly try to open the serial monitor until text displays. Past that I unfortunately don't have much advice to offer.

grezco, 2018-10-13:
Hi @facchinm 
With the script above, with arduino/ArduinoCore-samd#361 if I send a char by serial monitor, I freeze IDE (Wndows 10).

polygamma, 2018-10-13:
You may want to take a look at this: https://github.com/arduino-libraries/ArduinoLowPower/pull/9

grezco, 2018-10-14:
Hi @polygamma,
Nothing better with #9 for me. 
After wake up, in all case, I need to unplug / reconnected USB of MKR WiFi1010 device to be abble to see the serial message.

polygamma, 2018-10-14:
@grezco with the following example, you never see the second message?

```
#include "ArduinoLowPower.h" // https://www.arduino.cc/en/Reference/ArduinoLowPower
#include "RTCZero.h" // https://www.arduino.cc/en/Reference/RTC

/*
 * initial setup, will be run once at the start of the whole routine
 */
void setup() {
  // init serial connection
  Serial.begin(9600);
  while (!Serial) {}
  RTCZero().begin(false);
}

/*
 * will be run forever, after initial setup() call
 */
void loop() {
  Serial.println("I am being printed, because the device has not been sleeping, yet");
  
  // deep sleep of 5 seconds
  LowPower.sleep(1000 * 5);
  
  Serial.println("I am never being printed, because of a deadlock");
}
```

grezco, 2018-10-14:
@polygamma,
No I never see the second message.

polygamma, 2018-10-14:
Since you wrote in https://github.com/arduino-libraries/ArduinoLowPower/issues/7#issuecomment-425667428 that you are Windows 10, I tried the snippet with my fixes on Win10, too.
Sorry, have to confirm, that my changes to not help, if you are using Win 10.
Anyway: Using Arch Linux, and probably any Linux distribution, it does indeed fix the problems.

grezco, 2018-10-14:
@polygamma, thanks for the time spent trying to solve my problem.

Until I find a solution, I use 
````
  USBDevice.detach();`
  LowPower.sleep(1000 * 5);
  USBDevice.attach();
```
and reopen the serial monitor when the port appear.

facchinm, 2018-10-15:
@grezco @polygamma @durasno  thank you all for testing on different platforms; being able to reproduce the behaviour is crucial to get a solution.
@grezco are you testing with https://github.com/arduino-libraries/ArduinoLowPower/pull/8 applied? Because it implements (almost) the same solution you are using; I'd hate to insert OS-specific code based on the machine you are working with (these kind of solution never work in the real world)

grezco, 2018-10-15:
Hi @facchinm 
As previously said, #8 don't work (for me) with Windows10 and MKR1000.
With Arduino SAMD core - Pull Request #361, the serial port don't disappear in the device manager of Windows 10. But after wake up, in all case, I'm not abble to see the serial message.

facchinm, 2018-10-15:
@grezco so what did you mean with 
> Until I find a solution, I use ...

if you are unable to see the serial message? Does adding the extra lines change the behaviour over #8 or not?

grezco, 2018-10-15:
#8 don't change the behaviour when I use  USBDevice.detach();

`#include "ArduinoLowPower.h" // https://www.arduino.cc/en/Reference/ArduinoLowPower
#include "RTCZero.h" // https://www.arduino.cc/en/Reference/RTC

void setup() {
  // init serial connection
  Serial.begin(9600);
  while (!Serial) {}
  RTCZero().begin(false);
}

void loop() { 
  Serial.println("I am being printed, because the device has not been sleeping, yet");
  delay(10);
  USBDevice.detach();
  LowPower.sleep(1000 * 5); 
  USBDevice.attach();
  delay(5000);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);
  Serial.println("Wake up MCU");
}`

fabik111, 2019-01-10:
Hi guys, I had the same problem with MKR1000 and I've solved using the library version on GitHub (the version on Library Manager not works) and adding these lines of code on your project:
` USBDevice.detach();
  LowPower.deepSleep();
  USBDevice.attach();`

pchatill, 2019-03-20:
Hi.
Same problem for me: Serial not working after wakeup .
"USBDevice.detach(); LowPower.sleep(); USBDevice.attach();" helped me a lot,
But I need to close the sérial monitor and re-open (IDE 1.8.8 , windows 10).
And  I miss some informations on serial port if I don't do that quickly enough
Is there a way do to that automatically?
And is that planned to modify the library?
Thanks a lot for answers.
Patrick

facchinm, 2019-03-21:
Hi @pchatill , it was proposed for inclusion in the IDE (https://github.com/arduino/Arduino/pull/8046). That PR is already included in Beta builds of the IDE but has some problems with non arduino boards (esp8266 and similar) so we decided to postopone the merge in master.
Testing it would greatly speedup merge ;)

pchatill, 2019-03-21:
Thanks for your response facchinm.
I tried with IDE beta 1.8.9. 
It seems not necessary with this version to add :USBDevice.detach(); USBDevice.attach();
But it's always necessary to close serial monitor and reopen it.
The difference is that the serial monitor output is cleared when Arduino sleep.
The other difference is that the serial monitor looks active again when arduino wake-up (Autoscroll function active for example) but no data displayed ...

facchinm, 2019-03-21:
The beta should be numbered 1.9.0 , please double check if you are using that one :wink: 
Beside that, are you on Windows?

pchatill, 2019-03-21:
Sorry for the mistake: Yes I tried with Arduino 1.9.0-beta.
Yes windows 10

pchatill, 2019-06-09:
Hi facchim.
Do you have any news about the problem?
Thanks a lot...

netless-ww, 2020-01-01:
I have found the following code using the Arduino low power library is a partial solution - at least it gives you a delay to close & reopen the USB serial port, after the SAMD wakes up.

I'm using a 16x2 LCD to see what is happening while the serial port is unavailable - remove if you wish. Also the code is a bit messy with commented out sections where I have been trying other low power libraries.  

The method uses the USBDevice.detach() -- LowPower.deepSleep() -- USBDevice.attach() combination, together with a delay while(!Serial) {} to give you time to close & reopen the Serial port.

Note: delay(10000); after restarting the USBDevice is very important. The value of Serial is initially returned as 1, and will drop straight through the while delay - after the one second delay, the value of Serial changes back to 0 and is captured in the while loop until a new Serial window is opened and the value of Serial changes back to 1.

Still not an ideal solution, but at least you get a working Serial window back without loosing any output.

Incidentally, the LowPower.deepSleep();  library command also does a USBDevice.detach() -- USBDevice.attach() internally - without the external USBDevice.detach() -- USBDevice.attach() sequence, this will keep your existing Serial communication open (no need to restart) and can receive commands, but will not give any output to the serial monitor. Perhaps this is because there is no 1 second delay & the serial port being closed and reopened.

There are lots of funny timing issues going on here - I would appreciate it if anyone can shed more light on this. 

// **** INCLUDES *****
#include "ArduinoLowPower.h"
#include <Wire.h>
#include <LiquidCrystal_PCF8574.h>

LiquidCrystal_PCF8574 lcd(0x27);  // set the LCD address to 0x27 for a 16 chars and 2 line display

// External interrupt on pin 0 (use pin 0 to 24, except pin 4 on Arduino Zero)
const int pin = 12;
unsigned char count = 3;

void setup()
{
	// Wait for serial USB port to open
	Serial.begin(9600);
  delay(2000);
	Serial.println("***** ATSAMD21 Standby Mode Example *****");

  lcd.begin(16, 2); // initialize the lcd
  lcd.setBacklight(HIGH);
  printMessages(1, 0, 0);    //print  Found LCD
  delay(2000);               //small delay before the screen is cleared
  lcd.home(); lcd.clear();
	
	// ***** IMPORTANT *****
	// Delay is required to allow the USB interface to be active during
	// sketch upload process
	Serial.println("Entering standby mode in:");
	for (count; count > 0; count--)
	{
	  Serial.print(count);	
	  Serial.println(" s");
	  delay(1000);
  }
  // *********************
    
  // External interrupt on pin (example: press of an active low button)
  // A pullup resistor is used to hold the signal high when no button press
  pinMode(pin, INPUT_PULLUP);
  LowPower.attachInterruptWakeup(pin, blink, LOW);
}

void loop() 
{
	Serial.println("Entering standby mode.");
	Serial.println("Apply low signal to wake the processor.");
	Serial.println("Zzzz...");
  Serial.print("SCB_SCR_SLEEPDEEP_Msk  ");
  Serial.println(SCB_SCR_SLEEPDEEP_Msk, BIN);
  Serial.print("~SCB_SCR_SLEEPDEEP_Msk  ");
  Serial.println(~SCB_SCR_SLEEPDEEP_Msk, BIN);
  //Serial.println(GCLK->CLKCTRL.reg, BIN);
  printMessages(0, 0, 0);
  printMessages(0, 0, 1);
  
	// Detach USB interface
	   USBDevice.detach();
 
  // Enter standby mode
  // LowPower.standby();
  /* This is all LowPower.standby() does
  SCB->SCR |= SCB_SCR_SLEEPDEEP_Msk;
  __DSB();
  __WFI();
  */
  LowPower.deepSleep(5000); 
  
  //LowPower.idle(IDLE_2);
  lcd.setCursor(0, 0); lcd.print(F("Awake"));
   
  // Attach USB interface
     USBDevice.attach();
  // It appears the Serial is returned as 1 after the attach command
  // However it also appears to revert to 0 after 1 sec of inactivity
  // Thus being trapped in the following while statement until the new 
  // serial monitor is opened. If the following delay is omitted the
  // following while loop will pass straight through, without waiting for
  // the new serial monitor to open!!   
  
     delay(1000); 
  
  // Force delay until the old serial monitor is closed and a new one
  // is opened, changing Serial from 0 back to 1
  int s = Serial;
  lcd.setCursor(8, 0); lcd.print(s);
  while(!Serial) {lcd.setCursor(10, 0); lcd.print(s);s++;}
  lcd.setCursor(8, 1); lcd.print(Serial);
  lcd.setCursor(0, 1); lcd.print(F("Serial"));

  Serial.println("Awake!");
  Serial.println("Serial Communication Restored!");
  Serial.println("Send any character to enter standby mode again");

  // Wait for user response
  // Serial.available is 0 while there are no characters in the USB pipline,
  // causing the while to loop indefinitely.
  // As soon as a character is typed, Serial.available changes to 1 and 
  // characters are read from the USB pipline until it is empty and
  // Serial.available returns to 0 and the second while loop terminates.
  while(!Serial.available());
  while(Serial.available() > 0)
  {
		Serial.read();
	}

}

void blink(void)
{
 
}
/***************************************************************************/
//Print messages to the LCD

  void printMessages(uint8_t messNo, uint8_t messCol, uint8_t messRow ) {
  lcd.setCursor(messCol, messRow);                 //set up cursor possition

  if (messNo == 0)  lcd.print(F("                "));
  if (messNo == 1)  lcd.print(F("Found LCD       "));
}

AndreasWaldherr, 2020-10-15:
**Hi netless-ww**, 
thanks a lot for sharing your test results. 
Your workaround does the job also for my Arduino MKR WAN 1310 LoRa Module :-) 

Andreas

adamgarbo, 2021-04-17:
When using the `rtc.standby()` function from the RTCZero library and manually detaching/attaching the USBDevice, I am able to reliably reestablish communication with the serial port of a SAMD21-based microcontroller after sleep (Adafruit Feather M0). I am curious why this is not possible with the ArduinoLowPower's `LowPower.deepSleep()`, given the functionality is essentially the same. 

As a side note, the measured quiescent current draw of the `rtc.standby()` function is equivalent to `LowPower.deepSleep()`.

```
// Close serial port
Serial.end();   

// Safely detach USB prior to sleeping
USBDevice.detach();   

// Sleep until next alarm match
rtc.standbyMode(); 

// Re-attach USB
USBDevice.attach(); 

// Open serial port 
Serial.begin(115200);
```

Andreas1313, 2025-04-20:
At my configuration, I need `delay(1);` between `Serial.end();` and `USBDevice.detach();`. Otherwise it is not working, or the last "New line" is missing.

This is my test scetch:
```
//!!!Because of the sleep mode, upload the sketch by double-tapping the reset button and putting the board into the bootloader mode!!!
#include <RTCZero.h>
RTCZero rtc;

void setup(){
  Serial.begin(115200);
  while( ! Serial){}
  delay(1);
  Serial.println("setup");

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  rtc.begin();
  rtc.setTime(9, 10, 59);
  rtc.setDate(03, 04, 25);
  rtc.setAlarmSeconds(3);
  rtc.enableAlarm(rtc.MATCH_SS);
  //rtc.standbyMode();
}

void loop(){
  digitalWrite(LED_BUILTIN, ! digitalRead(LED_BUILTIN));
  USBDevice.attach();
  Serial.begin(115200);
  while( ! Serial){}
  Serial.print(rtc.getHours());
  Serial.print(":");
  Serial.print(rtc.getMinutes());
  Serial.print(":");
  Serial.println(rtc.getSeconds());
  Serial.end();
  delay(1); // !!!This is needed!!!
  USBDevice.detach();  
  rtc.standbyMode();
}
```
I use Win11,  MKR1010, IDE 2.3.5

gegy1, 2025-06-06:
I have something similar on macos.
When the MKR1010 Wifi goes to sleep and wakes up again, serial is not working. The IDE is also not finding the board anymore even not when double click the button to get the slow LED flashing. I need to restart my computer to get it working  again.