arduino-libraries/ArduinoLowPower #18: LowPower.Sleep twice as long as duration is set

caiello3, 2019-07-08:
Hello,

I am using the Arduino Low Power library with an MKR1000 board. I have found that when I implement the LowPower.sleep function the sleep cycle is lasting twice as long as the duration is set in the code. One other person has confirmed this in a separate sketch test. Please see the forum post with details below:

[https://forum.arduino.cc/index.php?topic=624855.0](url)

Thanks!

facchinm, 2019-07-08:
Hi @caiello3 ,
I've been able to reproduce it, thanks. The problem looks related with the serial port status.
In [this](https://github.com/arduino-libraries/ArduinoLowPower/commit/8cf4c737ddb755e76c20407df3d70cc507bd9651#diff-a3e4eb55ed87714668dd193a2024a581R20) line we check if we need to deactivate and restore the Serial port, only if it has previously been connected. 
The issue seems related with that check, since opening the USB serial port fixes the delay.
I'll investigate the issue a bit more and will let you know!

caiello3, 2019-07-08:
Great. Thanks for taking a look and confirming. Looking forward to an update.

facchinm, 2019-07-30:
Hi @caiello3 ,
After rechecking all the code I can confirm it's working as expected: the timings are off by 1 second at most, depending on the USB status; this is why the 2 seconds test seems to confirm the issue :slightly_smiling_face: Probably the double period you are noticing is due to a "ping" out of 2 failing for some other reasons.
I'd leave the issue open since the 1 second off is a bit problematic for shorter sleeps and should be solved in the future, but longer sleeps are ok :slightly_smiling_face:

jsiddall, 2021-03-21:
I don't think the serial port status is affecting this.  I tried running the sleep function without any of the USB related lines and even with sleep(1), which should be 1 millisecond, I still get a sleep of 1 second or so.  It appears that it may just take that long to restart the 48 MHz clock after shutting it down.

garfieldand, 2021-06-08:
@facchinm, This problem is occurring even when there serial port is not in use at all and never opened.  Confirmed with power analyzer.  Seen in sleep seconds >1 second as well.

KravitzLab, 2023-07-26:
Hi!  Is there any more information on this?  I'm experiencing the same issue with ArduinoLowPower v1.2.2 with an [Adafruit Adalogger M0 board](https://www.adafruit.com/product/2796).  I am using the "TimedWakeup" example in the library, and it always sleeps for ~1s longer than I want.  ie: if I modify the sleep call to:
`  LowPower.sleep(1);`

I see this power consumption, showing the processor sleeps for ~934ms, not 1ms.   
[image]

Is there any more information on why this happens or how to get sub-1s sleep times with this library?  In our application we'd like to poll a sensor 5x per second, and sleep between polling.  I've also tried detaching USB before sleep but it did not change this behavior. Thanks!

KravitzLab, 2023-07-26:
Just to throw out a bit more information, if I include the line `PM->SLEEP.bit.IDLE = 1;` before `LowPower.sleep(1)`, the M0 sleeps for ~450ms, so that's an improvement (and maybe a clue to what's going on?)
[image]

I also found that `LowPower.idle(1) ` causes the M0 to enter the idle state for only ~50ms.  This is not that useful because idle still burns ~6mA of current, but the timing is much faster.
[image]