arduino-libraries/ArduinoLowPower #19: Can't get it to low power during sleep - Samd21

maxholgasson, 2019-09-29:
Hi,
I'm using a ATSAMD21G18A with this library and while sleeping the power consumption doesn't go 2mA. Unfortunatly that's to much for my application. Is there any chance to get it lower with this lib?

facchinm, 2019-09-30:
Hi @maxholgasson ,
do you have anything connected to the chip? All pins should be high Z to reach the maximum powersave, so you can skip pin initialization by declaring VERY_LOW_POWER at board level

https://github.com/arduino/ArduinoCore-samd/blob/b0eb400803cadf9727ab38d5b1c2fb46ab8eeec9/cores/arduino/wiring.c#L85-L91

https://github.com/arduino/ArduinoCore-samd/blob/master/boards.txt#L219

maxholgasson, 2019-10-01:
Thanks for your answer!!
Yes I had some other parts connected but tried it also with the blank chip, only a crystal attached
I will try your suggestion