arduino-libraries/ArduinoLowPower #21: ArduinoLowPower working in all nRF52 architectures?

morettigiorgio, 2019-11-29:
I am using ArduinoLowPower library on my Arduino Nano 33 BLE but during compile a "WARNING: library ArduinoLowPower-master claims to run on samd, nrf52 architecture(s) and may be incompatible with your current board which runs on mbed architecture(s)." is shown and then it is not compiled.
I have a doubt... Arduino Nano 33 BLE is a Nordic nRF52 !!, I have to specify this in my ino file?

Thanks

facchinm, 2019-11-29:
The `nrf52` target refers to the Arduino core for Primo board; NANO 33 BLE is based on `mbed` core and it automatically handles low power (simply calling `delay()` will trigger the low power routines).
Hope it is enough, if you have any other question feel free to ask here :wink:

closed 2019-11-29 by facchinm

morettigiorgio, 2019-11-29:
I will never finish to learn. Thank you a lot

Giorgio