arduino-libraries/ArduinoLowPower #37: LowPower.sleep() on Arduino MKR WiFi 1010

gsdevme, 2021-01-14:
Hi,

Trying to use this library on the Arduino MKR WiFi 1010 seems to ignore the sleep and/or just goes into a frozen state and never wakes up

```c
#include <ArduinoLowPower.h> // https://www.arduino.cc/en/Reference/ArduinoLowPower

void setup() {

}

void loop() {
  Serial.begin(9600);
  while (!Serial);
  
  Serial.println("Waiting 10 seconds");
  
  delay(10000);

  Serial.println("Sleeping");
  Serial.flush();

  LowPower.sleep(5000); // sleep for 5 seconds
}
```

[image]

MattStarfield, 2021-01-19:
When the MKR1010 goes to sleep, it will lose the Serial connection. This is expected behavior. In your sketch, when the device wakes up after `LowPower.sleep(5000); // sleep for 5 seconds`, the code will return to the top of `loop()` where you have it waiting for the serial connection `while (!Serial);`. Since the Serial connection was cut off at `sleep`, your code will hang here.

Try putting your `while (!Serial);` in `setup()` and instead of `loop()` printing to the Serial Monitor, have it toggle an LED each loop.

gsdevme, 2021-01-21:
Interesting, I think I maybe adapted this from an ESP32 which perhaps keeps the serial connection open or at least implicitly reconnects in the background. I can confirm a blink is working as expected. Will mark as resolved thanks for that.

closed 2021-01-21 by gsdevme