arduino-libraries/ArduinoLowPower #40: ArduinoLowPower and RTCZERO.h

jot4p, 2021-03-15:
Hi,

I need to use RTCZero to create a daily alarm based on RTC time, but I need to user ArduinoLowPower lib to create a short sleep of 10 seconds in loop.

Why do not work ?

#include <RTCZero.h>
#include "ArduinoLowPower.h"

RTCZero rtcZero;

const byte rtcZero_seconds = 0;
const byte rtcZero_minutes = 22;
const byte rtcZero_hours = 14;
const byte rtcZero_day = 15;
const byte rtcZero_month = 3;
const byte rtcZero_year = 21;

void setup() {
  
  Serial.begin(9600);

  rtcZero.begin(); // initialize RTC 24H format
  rtcZero.setTime(rtcZero_hours, rtcZero_minutes, rtcZero_seconds);
  rtcZero.setDate(rtcZero_day, rtcZero_month, rtcZero_year);
  rtcZero.setAlarmTime(14, 22, 30);
  rtcZero.enableAlarm(rtcZero.MATCH_HHMMSS);
  
  rtcZero.attachInterrupt(alarmMatch);

}

void alarmMatch()
{
  Serial.println("Alarm Match!");
  
}

void loop() {
  
 //Serial.println("TESTE");
 LowPower.sleep(2000);
 delay(1000);

}
Thanks.

per1234, 2021-03-16:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

---
Related: https://github.com/arduino-libraries/RTCZero/issues/70

closed 2021-03-16 by per1234

jot4p, 2021-03-16:
this is a bug, you lock access to RTCzero, to define alarme based on hours.