arduino-libraries/ArduinoLowPower #53: Compile problem in Arduino2.0 [nrf_lpcomp_input_t does not name a type] and more

eyesblue, 2022-10-18:
The compiler shows me the problem below, I used Arduino 2.0, and I have no idea what's happening, please help.

```
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:31:1: error: 'nrf_lpcomp_input_t' does not name a type; did you mean 'nrf_gpio_pin_input_t'?
   31 | nrf_lpcomp_input_t aPin[]={NRF_LPCOMP_INPUT_1, NRF_LPCOMP_INPUT_2, NRF_LPCOMP_INPUT_4, NRF_LPCOMP_INPUT_5, NRF_LPCOMP_INPUT_6, NRF_LPCOMP_INPUT_7};
      | ^~~~~~~~~~~~~~~~~~
      | nrf_gpio_pin_input_t
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp: In member function 'void ArduinoLowPowerClass::enableWakeupFrom(wakeup_reason, uint32_t, uint32_t, uint32_t)':
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:111:3: error: 'detect_mode' was not declared in this scope
  111 |   detect_mode mode;
      |   ^~~~~~~~~~~
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:112:16: error: 'DOWN' was not declared in this scope
  112 |   if(option == DOWN)
      |                ^~~~
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:113:4: error: 'mode' was not declared in this scope; did you mean 'modf'?
  113 |    mode = DOWN;
      |    ^~~~
      |    modf
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:114:21: error: 'UP' was not declared in this scope
  114 |   else if(option == UP)
      |                     ^~
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:115:4: error: 'mode' was not declared in this scope; did you mean 'modf'?
  115 |    mode = UP;
      |    ^~~~
      |    modf
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:117:4: error: 'mode' was not declared in this scope; did you mean 'modf'?
  117 |    mode = CROSS;
      |    ^~~~
      |    modf
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:117:11: error: 'CROSS' was not declared in this scope
  117 |    mode = CROSS;
      |           ^~~~~
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:118:3: error: 'nrf_lpcomp_config_t' was not declared in this scope; did you mean 'nrf_pwm_configure'?
  118 |   nrf_lpcomp_config_t config={(nrf_lpcomp_ref_t)event, (nrf_lpcomp_detect_t)mode};
      |   ^~~~~~~~~~~~~~~~~~~
      |   nrf_pwm_configure
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:119:25: error: 'config' was not declared in this scope
  119 |   nrf_lpcomp_configure(&config);
      |                         ^~~~~~
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:119:3: error: 'nrf_lpcomp_configure' was not declared in this scope; did you mean 'nrf_pwm_configure'?
  119 |   nrf_lpcomp_configure(&config);
      |   ^~~~~~~~~~~~~~~~~~~~
      |   nrf_pwm_configure
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:122:27: error: 'aPin' was not declared in this scope; did you mean 'asin'?
  122 |   nrf_lpcomp_input_select(aPin[pin-14]);
      |                           ^~~~
      |                           asin
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:122:3: error: 'nrf_lpcomp_input_select' was not declared in this scope
  122 |   nrf_lpcomp_input_select(aPin[pin-14]);
      |   ^~~~~~~~~~~~~~~~~~~~~~~
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:123:3: error: 'nrf_lpcomp_enable' was not declared in this scope; did you mean 'nrf_pwm_enable'?
  123 |   nrf_lpcomp_enable();
      |   ^~~~~~~~~~~~~~~~~
      |   nrf_pwm_enable
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:124:27: error: 'NRF_LPCOMP_TASK_START' was not declared in this scope; did you mean 'NRF_RTC_TASK_START'?
  124 |   nrf_lpcomp_task_trigger(NRF_LPCOMP_TASK_START);
      |                           ^~~~~~~~~~~~~~~~~~~~~
      |                           NRF_RTC_TASK_START
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:124:3: error: 'nrf_lpcomp_task_trigger' was not declared in this scope; did you mean 'nrf_pwm_task_trigger'?
  124 |   nrf_lpcomp_task_trigger(NRF_LPCOMP_TASK_START);
      |   ^~~~~~~~~~~~~~~~~~~~~~~
      |   nrf_pwm_task_trigger
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:125:33: error: 'NRF_LPCOMP_EVENT_READY' was not declared in this scope
  125 |   while(!nrf_lpcomp_event_check(NRF_LPCOMP_EVENT_READY));
      |                                 ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:125:10: error: 'nrf_lpcomp_event_check' was not declared in this scope; did you mean 'nrf_pwm_event_check'?
  125 |   while(!nrf_lpcomp_event_check(NRF_LPCOMP_EVENT_READY));
      |          ^~~~~~~~~~~~~~~~~~~~~~
      |          nrf_pwm_event_check
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:132:29: error: 'g_APinDescription' was not declared in this scope
  132 |    nrf_gpio_cfg_sense_input(g_APinDescription[pin].ulPin, NRF_GPIO_PIN_PULLUP, NRF_GPIO_PIN_SENSE_LOW);
      |                             ^~~~~~~~~~~~~~~~~
c:\Users\eyesblue\SyncFolder\Coding\Arduino\libraries\Arduino_Low_Power\src\nrf52\ArduinoLowPower.cpp:134:29: error: 'g_APinDescription' was not declared in this scope
  134 |    nrf_gpio_cfg_sense_input(g_APinDescription[pin].ulPin, NRF_GPIO_PIN_PULLDOWN, NRF_GPIO_PIN_SENSE_HIGH);
      |                             ^~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

per1234, 2022-10-18:
Hi @eyesblue. Which board did you have selected in the Arduino IDE when you encountered this error?

eyesblue, 2022-10-19:
Thanks for your reply, my hardware is Adafruit nRF52840 Feather, it also happens on arduino-1.8.19, it always happen if 

the #include "ArduinoLowPower.h"  line exist.

per1234, 2022-10-19:
This is the problem. That board is not supported by the library. As explained here, the only nRF52-based board that is supported is the retired Arduino Primo: https://github.com/arduino-libraries/ArduinoLowPower/issues/3

You will either need to find another library to use with your board, or do without a library and use whatever sleep capabilities are available from the core or toolchain directly.

closed 2022-10-19 by per1234

eyesblue, 2022-10-19:
thanks for your help, I have a suggestion for you that remove the description "support nRF52-base board", I think the real problem is the definition of "nRF52", the nRF52-base should contain the nRF52 series, including nRF52840, nRF52833, nRF52832, nRF52820,  nRF52811, nRF52810, nRF52805..., but the library just support nRF52832, if there still have the descript that "support nRF52-base board", I think such question will happen again and again.

sorry for wasting your time with my mistake, and thanks for your help, good luck, and have a nice day.